elewin/pi-weather-station #2: Different Monitor size optimization

glipets, 2020-09-10:
Works well, but attempting to use your project on 3.5 TFT screen, and the chromium window doesn't look good, basically can only see half the output. Can anything be done to optimize it for smaller screens ?

elewin, 2020-09-10:
It would mostly be a matter of changing the CSS up to make adjustments based off of a media query breakpoint; I'll see if I have time over the next couple weekends to play around with it and see what I can come up with (one of the big challenges will be the limited screen real estate of one of those displays). No promises, but I'll try to look into it!

mlcampbe, 2024-01-04:
Is there any update here? For example, I am playing around with trying to get the output to display in an iFrame that is a MagicMirror module. That is working properly but the width of the output is being cut off on the right hand side. See the below screenshot of my TV. In my case the TV is reporting the size as 1360x768

[image]

Is there a key to being able to resize the app somehow? I am open to making the entire thing slightly smaller or even resizing the infopanel on the right to be a little larger. 

Any thoughts or simple ideas to accomplish this?

mlcampbe, 2024-01-05:
I solved my own problem. I found the min-width=800 in the App styles.css. I changed this to 700 and now it displays properly.