elewin/pi-weather-station #5: Are you going to upgrade the climacell API calls

mlcampbe, 2020-12-28:
I use climacell for a personal project as well and got an email that they are moving from v3 of the API to v4 in the next 6 months or so. Please tell me you are planning on upgrading this project to use the new v4 calls so that this will continue to work.

elewin, 2020-12-28:
I'll take a look at it!

bakercp, 2020-12-28:
Just to chime in here. I generated a new account on climacell and learned from tech support that the newly generated api keys are "v4" keys and don't work with v3 calls, which is why I was unable to get pi-weather-station to work. Excited to try out v4 calls! Thanks @elewin

bakercp, 2020-12-28:
I did a quick dive into it. On the API side, several field names have changed and sunrise/sunset is no longer available on the "realtime" "1m" endpoint.  These minor changes got the api working, but I didn't have time to update the react properties, etc.

https://gist.github.com/bakercp/86003bfdfdd361b7e2ae59ad7f0097c4

bakercp, 2020-12-28:
Here is a PR as it's easier to see the diff https://github.com/elewin/pi-weather-station/pull/6/commits/85f3c2508deee8110db7c2a3820442958d4214d1

elewin, 2020-12-28:
Thanks for checking that out, I'll try to get around to it this weekend. It doesn't seem like it should be too complicated.

bakercp, 2020-12-28:
Thanks for your work on this! Cool project :)

webeling67, 2021-01-10:
Looks like three files are affected by the V4 change:
bundle.min.js and Appcontext.js have to get new API calls
index.js and Appcontext.js need to deal with the new response

Up to now I only recognized new API calls and new responses. Anything that was removed and is needed?

elewin, 2021-01-23:
I've created a branch that uses ClimaCell API v4:
https://github.com/elewin/pi-weather-station/tree/climacellv4

If anyone wants to test it out, let me know! I'll play around with it myself for a bit and merge it if it looks ok

mlcampbe, 2021-01-25:
Seems to be working just fine for me. No problems at all that I can find.

chrism583, 2021-01-27:
Loving the project!  I've been using it flawlessly for the last month or so.  What's the upgrade path to v2?  Do we just re-clone the repo and restart?

elewin, 2021-01-29:
ClimaCell v4 support added https://github.com/elewin/pi-weather-station/pull/8

elewin, 2021-01-29:
> Loving the project! I've been using it flawlessly for the last month or so. What's the upgrade path to v2? Do we just re-clone the repo and restart?

You could do that, or pull the latest changes with `git pull origin master`

If you decide to re-clone instead, you may want to backup your `settings.json` file so you won't have to re-enter your keys again (Though you'll need to update the weather key with a new ClimaCell v4 key)

closed 2021-01-29 by elewin