elewin/pi-weather-station #7: Add option to adjust temp locally

mlcampbe, 2021-01-01:
I'm not sure where climacell gets its temp from but suspect it is noaa or airports or local new stations. However it does not necessarily match with my backyard. It seems that the reported value is about 3-4 degrees cooler than my backyard gauge. So, is it possible to add a setting to add/subtract from the climacell temp so that we can make the value closer to our actual temps? For example, I'd like to add +3 degrees to the value reported from climacell in my case.

ragd011, 2021-08-19:
You can tweak your install by modifying the code and rebuilding it.  I just tested it with mine.

edit the file in client\src\components\CurrentWeather\index.js
after line 52 add aline like this:
const teamperatureAdj = temperature + 3;

Then on line 57, update it from temperature to temperatureAdj

Then on the command line run npm install webpack

After that installs cd into the pi-weather-station/client/ directory and run "npm run prod"

This will rebuild the package with your changes.  
Then cd ..
and run npm start
Your temp should be increased by 3 degrees.

mlcampbe, 2021-08-19:
Yes I figured out how to hardcode a temp adjustment just fine and that worked. But was looking for a setting that I could use to adjust the temp without having to recompile and restart everything. I have been able to add that as well but it is not quite user-friendly as of yet. On the settings page I added a new text box in the middle of the page. That pushed the rest of the page down just enough that the save button is off the bottom of the page and thus I can't click it. There is no scroll bar either so I am looking for a way to fix that then it would be usable. For now I can edit the settings.json file and restart.