elewin/pi-weather-station #9: Stablility issues

savilan77, 2021-01-30:
First, thank you for the  great code. Mine works well in the beginning, but after some hours of use the webserver crashes and stops responding.  I have the same issue with two different raspberry boards. Any tips how to solve the issue?

STFUandFish, 2021-02-07:
What hardware is this aimed at? I ran hamweather for years on a 3 A+ with zero problems, this thing drives it into the ground.

I tried a 3 B+, same problem.

STFUandFish, 2021-02-07:
Update: Runs on 3 B+. Had to weak down the GPU RAM. I don't know why, but at higher RAM settings it forced pretty much everything to swap.

savilan77, 2021-02-09:
I am running on Raspberry 4. I guess I have solved my case. I run Pi Weather Station with nodemon and it has been stable last week or so.

closed 2021-02-09 by savilan77