elewin/pi-weather-station #10: Incorporating Local Data

kjaske, 2021-02-18:
Any interest in making it possible to incorporate data from backyard sensors--whether by replacing or simply by supplementing the temp / humidity / wind data from ClimaCell? Having the radar, the sunrise/sunset times, and the beautiful layout are super. It'd be even better to be able to incorporate local data as well. So far, my own efforts to that end have proven fruitless.

elewin, 2021-02-22:
Where would the data be coming from? In what format would it come in, and how would it be accessed?

kjaske, 2021-02-22:
It would be coming from a local MariaDB, which stores my backyard weather readings in a single SQL table. I was hoping--ignorantly, and thus probably vainly--that temp, humidity, pressure, wind speed, wind direction, and air quality could be pulled periodically from the local DB and slotted in in place of the more generic ClimaCell data. Having played around with the pi-weather-station files, though, it seems not so simple… at least for this newbie. 😊

elewin, 2021-02-22:
It sounds like that would take a pretty significant re-write to do. Basically the node server would need to connect to your database directly, and then make queries against it that are accessible to the webpage (the app you see running) through some local rest endpoints, and then you'd have to have the app shape the data it gets back to match what its normally expecting from ClimaCell

closed 2021-02-22 by elewin