elewin/pi-weather-station #11: full screen ?

Eric738, 2021-02-20:
Hello
and thank you for your beautiful project that I managed to reproduce although not a linux pro. Just how to get full screen and do at startup raspi boot on your application? thank you in advance for your info
eric
[image]

mlcampbe, 2021-02-22:
Take a look at the reddit thread at https://www.reddit.com/r/raspberry_pi/comments/il03rk/i_made_weather_station_app_and_designed_it_for_a/ and you will get some pointers. The general directions are:

Step 1. Comment out the line in /home/pi/pi-weather-station/server/server.js that says:

//  await open(`http://localhost:${PORT}`);
Save that file.

Then create a file, /home/pi/.config/lxsession/LXDE-pi/autostart

touch /home/pi/.config/lxsession/LXDE-pi/autostart
Edit with your favorite editor and put in code similar to the following:

@xset s noblank
@xset s off
@xset -dpms
@bash /home/pi/weather.sh

Then create the referenced /home/pi/weather.sh and add the following:

#!/bin/bash
cd ~/pi-weather-station/
/usr/local/bin/npm start ~/pi-weather-station/server/index.js &
chromium-browser --noerrdialogs --disable-inforbars --incognito --kiosk http://localhost:8000/&

Then chmod +x weather.sh to make it executable. Reboot, and your display will pop up Chromium in kiosk mode, incognito, with no errors. I did get the upgrade dialog though so I need to find a way to disable that.

I found that when the pi boots chromium starts up but initially gets a page not found error. That goes away about 10 sec later automatically though once the npm server starts up.

Eric738, 2021-02-23:
thank for your answer but in /home/pi/pi-weather-station/server/ i have only 3 files : index, geolocationCtrl and settingsCtrl.js , not server.js  idea ? i cant go to the next steps

mlcampbe, 2021-02-23:
Sorry, I copied from the reddit page. The file to edit is the index.js not the server.js (at least in the current version).

Eric738, 2021-02-23:
thank a lot , i check !

mlcampbe, 2021-03-01:
Create the .config directory manually if it does not already exist.