elewin/pi-weather-station #14: Upgrading

Chappie88, 2021-03-01:
Sorry, a rookie when it comes to RPI commands. Is there a way to upgrade to v2 without starting over? I have tried the sudo apt-get upgrade with no luck.

chrism583, 2021-04-09:
I came from a similar place as you with RPI commands but I was able to successfully update following these steps:
SSH into your pi, cd into your pi-weather-station folder, and run "sudo git pull origin master".  Initially I got commit errors but was able to resolve those by running "sudo git reset --hard" and then re-running "sudo git pull origin master".  Remember to register for a ClimaCell v4 API because you'll need to enter that new key after the update.

elewin, 2021-04-15:
You're going to have to pull down the latest version of the repo.

closed 2021-04-15 by elewin

Chappie88, 2021-04-15:
Have you noticed any advantages with the newer software setup that makes the upgrade worth it?

chrism583, 2021-04-15:
The real driving force is that later this year ClimaCell will shutdown their v3 API.