elewin/pi-weather-station #15: NOM looking for package.json

Rot10Crotch, 2021-05-26:
Thank you for taking the time & effort to make this. Unfortunately, I am unable  to get it running. First off I have node V14.17 installed. when I issue the command npm start, I Get the following response:
npm ERR! syscall open
npm ERR! path /home/pi/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-05-26T16_13_08_108Z-debug.log
[2021-05-26T16_13_08_108Z-debug.log](https://github.com/elewin/pi-weather-station/files/6548407/2021-05-26T16_13_08_108Z-debug.log)

It seems it is looking for package.json. I am attaching the log file

ragd011, 2021-08-18:
I just installed this myself on an RPI with touchscreen.  When you clone from Git, you should have a directory called pi-weather-station.  You want to cd into that and then run npm start.  That's where the package.json is to get it started I believe.

elewin, 2021-09-02:
@ragd011 Is exactly right, after cloning the project you have to go into the directory it creates before running `npm install`

closed 2021-09-02 by elewin