elewin/pi-weather-station #20: Adjustments to weather API call rates

dpmcintyre, 2021-09-10:
Tomorrow.io/Climacell seems to limit my free account to 100 API calls a day, not 500 like they advertise. With no channels to contact them since I don't pay them.
I run out of API calls by morning. 

Can there be an adjustment setting to how often the weather API is called? 
Are there any other alternatives out there after Tomorrow.io took over?

elewin, 2021-09-10:
There isn't an adjustment that you can easily make in the app itself, but if you don't mind messing with the code and rebuilding the app with webpack, the update intervals are defined here:
https://github.com/elewin/pi-weather-station/blob/master/client/src/components/WeatherInfo/index.js#L10

Currently it updates the current weather every 10 minutes, the hourly weather every hour, and the daily weather every 24 hours. In total that should be 156 calls / day. Changing the daily call interval to 20 minutes should get you under 100.

Keep in mind though that this is if you keep the map location stationary, as every time you move the map to a new area it will make a new query for each one of those as it gets the data for the new coordinates you scrolled to.

elewin, 2021-09-13:
I'll see about updating the app to make the current weather data interval adjustable from within the settings page to make it easier for people.

morphias2004, 2022-05-20:
Hi @elewin ,

Love this project! 👍 

Just wondering if you ever went any further with this?

I'm happy with the Info, Hourly and Daily updates, but would like to reduce the rain chart updates from 10mins to 5mins.

I found the index.js file in `pi-weather-station\client\src\components\WeatherMap` and edited L10 from 10 minutes to 5 minute and then rebooted, but it made no difference.

`const CURRENT_WEATHER_DATA_UPDATE_INTERVAL = 5 * 60 * 1000; //every 5 minutes`

I'm not familiar with webpack.

I did successfully change the conversion, Settings button and displayed units for the wind speed from mph to knots by editing a few `index.js` files and the `bundle.min.js` file.  :) 

Thanks in advance...

elewin, 2022-05-24:
If you run webpack from the commandline, it will read the updates you made in the source files and generate a new bundle.min.js file for you, but it looks like you managed to manually do it. Looks great!