elewin/pi-weather-station #22: auto start the Weather-Station

DennisCor, 2021-12-01:
I try my best, but I can't get your program to start automatically.
I wrote a bash-script (start_weather.sh) with the following call: 
`#!/bin/bash
cd ~/pi-weather-station/
npm start `
and made it executable.
`chmod +x /home/pi/start_weather.sh`
If I enter:
` $ /home/pi/start_weather.sh`
it starts like it should.
But if I put it in a crontab nothing hapend.
`0 9 * * * /home/pi/start_weather.sh`

Try the same with your bash https://github.com/elewin/pi-weather-station/issues/12 (just without chromium):
#!/bin/bash`
cd ~/pi-weather-station/
/usr/local/bin/npm start ~/pi-weather-station/server/index.js`
unfortunately without success

Do you have an idea where the problem could be or what I can still try ?

k5map, 2022-05-31:
I was able to get mine to start automatically during boot using crontab

patrickdickey52761, 2022-06-15:
Try changing your crontab entry to @reboot instead of the 0 9 ***. That way it starts on reboot and stays running. What you have now is every 9 minutes it tries to start whether it's running or not.

@reboot /home/pi/start_weather.sh

carl1961, 2023-07-29:
@k5map  can you share your crontab  commands?

k5map, 2023-07-30:
@carl1961 ...  Definitely; here are the steps to get the app to start on boot:

* crontab -e
* enter this at the bottom —  @reboot bash /home/pi/weather-start.sh  (the name of the bash script can be anything you want)

Here is what I have in my script in my home directory:
* cd $HOME/pi-weather-station/
* npm start &

carl1961, 2023-07-30:
Thanks @k5map  another question is what is your remedy to get full screen when it starts up. I am setting up this for my wife as a 7 inch screen with no key board and mouse. so if power fail reboot it boots to full screen

k5map, 2023-07-30:
@carl1961 ... I don't know the answer to your question.  For my setup, I'm using the app on a RPi and accessing the webpage remotely from my different computers.  If you Google Pi Weather Station, I think you'll find multiple YouTube videos which might give you the information to make that setting.

carl1961, 2023-07-30:
Thanks @k5map  thats what I been finding , found this and going to try

https://github.com/xieby1/fullscreen

carl1961, 2023-07-30:
@k5map Got it all working, this is what I had t do. I did everything you told me,  could not get browser to pop up and learned you cannot start browser through crontab    https://forums.raspberrypi.com//viewtopic.php?t=231987
did what this link showed for chromium-brower to start full screen https://raspberrypi.stackexchange.com/questions/69204/open-chromium-full-screen-on-start-up

what I put at bottom of  autostart file        sudo nano /etc/xdg/lxsession/LXDE-pi/autostart

@xset s off
@xset -dpms
@xset s noblank
@chromium-browser --start-fullscreen http://localhost:8080

chrontab -e    starts the npm start  as you told in your post at boot
 /etc/xdg/lxsession/LXDE-pi/autostart starts chromium-browser at boot