elewin/pi-weather-station #23: Opeweather option?

Pshemas, 2022-03-11:
I wonder whether there's easy way to switch to Openweather? ClimaCell seems to be no more, but I'd still love to use this project.

mlcampbe, 2022-03-16:
I received an email from Climacell recently that said it detected that I was using the URL data.climacell.co/v4/timelines and that I needed to transition to api.tomorrow.io. I was able to modify the project to simply replace data.climacell.co with api.tomorrow.io and all is well. I believe that I only made changes in the client/src/AppContext.js file (several references were modified in the file) and then rebuilt the dist/bundle.min.js using:

cd client
npm run-script prod

(If you are using nodejs v17 then it will fail with a "hash" error. You can get around this by setting "export NODE_OPTIONS=--openssl-legacy-provider" before runining the npm command.

I hope that helps.

k5map, 2022-05-31:
This is a great app... I use OpenWeather for several of my other projects, so I would prefer to have to option of using it for this as well (few API keys to manage) but happy with ClimaCell so far.

gdaysoul, 2024-05-08:
Hi I received this message a while ago and I have tried to find the location where I can make the change and it sounds similar to what [mlcambe] (https://github.com/mlcampbe) mentioned: 
"We're reaching out with an important update regarding your API services with Tomorrow.io.
As part of our ongoing efforts to improve service quality, integrity, and security, we are migrating our API domain. 
Upon reviewing our records, we noticed that your account is currently utilizing the old API domain.
To avoid any interruption in service, we kindly ask you to update your API endpoint from [data.climacell.co](http://data.climacell.co/) to  [api.tomorrow.io](http://api.tomorrow.io/)  by May 11."

Any ideas please? I am very much a beginner but can follow directions I think! :-)