elewin/pi-weather-station #27: ClimaCell redirects to tomorrow.io, and does not seem to be allowing public (free) connections

GreatLakesHam, 2022-05-30:
Just curious if I'm missing something, or not doing something correct?

I'm trying to install and setup  the pi weather station for the first time.  I have the application running, and the config screen to enter the API keys.  I have the "Maps API key", and the "Geolocation API Key".  However, I am making no progress on the Weather API key.

The directions I've seen and on this site have links to ClimaCell (which redirects to tomorrow.io) and then I try to sign up, I get "enter a business email address".  If I add emails from yahoo, gmail (and other carriers), I get "sorry that's not a business email".  Can someone give me a tip on how to get an account created?  Or maybe there is an alternate weather API site?

Thanks,
GreatLakesHam

patrickdickey52761, 2022-06-12:
If you go to https://www.tomorrow.io/weather-api/ you can sign up with a personal email and get free access. You're limited to a certain number of API calls, if I remember correctly.

bm3719, 2022-08-04:
Maybe someone else is seeing this too:

tomorrow.io won't let me log in anymore with my old user/pass.  If I try to create a new account with the old email, it says an account already exists.

D33pfield, 2023-01-02:
> If you go to https://www.tomorrow.io/weather-api/ you can sign up with a personal email and get free access. You're limited to a certain number of API calls, if I remember correctly.

This is not correct. Your link still redirects to the same sign up page that requires a "work" email.