elewin/pi-weather-station #28: change default viewing radius?

pbhound, 2022-06-04:
is there a setting to use to change the default viewing radius? I live in a metropolitan area with several cities all right next to each other and within about a 20-mile radius of my address. but the map shows what appears to be about a 50-mile radius. Is there a setting that I can use to change this?

mlcampbe, 2022-07-08:
I've not tried it but perhaps look at changing the zoom level. In the file https://github.com/elewin/pi-weather-station/blob/7f306e8a8312e764a411b952f8e25375af2d7a43/client/src/components/App/index.js#L46 is the default zoom level of 9. I don't know if you should go larger or smaller but you can change that then rebuild the client parts:

cd client
npm install
npm run-script prod

See if that helps

bnelson333, 2023-06-16:
can confirm that this works, thank you! I wanted to zoom in further than the default (I was seeing too many cities that are too far away from me). I changed my zoom level to 12, rebuilt as you suggested, and it works great.