elewin/pi-weather-station #30: Unable to retrieve map data

k5map, 2022-06-16:
I've had my pi-weather-station up and running for almost a month but earlier today, my RPi stopped being able to download the map from Mapbox.  I waited 8 hours to see if the problem would resolve itself but it did not.  I then checked the Mapbox website and the API key was still valid and showed no problems.  To confirm the API key was not the issue, I had Mapbox issue a new key.  After entering this new key into the RPi, it is still unable to download the map.  Anyone have any ideas how to fix this?

lerilar, 2022-06-22:
Same here, this however shows data:
https://api.mapbox.com/styles/v1/mapbox/streets-v11/static/10.5977,59.7777,9.3,0/800x480?access_token=<yourkey>

I have also noticed two orange/yellow shapes that seems to be stuck on the weather map. You can see them on https://www.rainviewer.com/weather-radar-map-live.html using the same coordinates. (Svelvik, south of Oslo, Norway)
[image]

k5map, 2022-07-07:
My issue was resolved by making the change outlined in Issue #32

closed 2022-07-07 by k5map