elewin/pi-weather-station #31: Cannot retrieve map data

LuxStatera, 2022-06-24:
I have been running the weather radar for a while now but recently had a power outage. When I restarted my pi and the weather radar it brings me to the same localhost page but the map is not there or any weather data. It just says "Cannot retrieve map data. Did you enter an API key?". I double checked my API keys and even tried new ones but I still get the same error. Any ideas on how to fix this?

k5map, 2022-06-27:
I'm having the same error but have not been able to identify the cause.  I've obtained new keys and it did not resolve the issue.

RNET1, 2022-07-13:
Be sure you have entered the LAT/LON values. This seems to break the code somewhere, for some, when not populated. One person stated they had to trim to 4 decimal places as opposed to 6. xx.xxxx

A reboot may be needed too.

LuxStatera, 2022-08-27:
Thank you! This fixed it for me

k5map, 2022-08-27:
Removing my Geolocating key and manually entering the LAT/LON (5 digits worked for me) resolved this issue for me as well