elewin/pi-weather-station #32: Stopped retrieving weather data

halifax57, 2022-07-06:
My system was running continuously and perfectly for several months, and about 3-4 days ago it stopped retrieving weather data. Map and radar and time/sunrise/sunset are all good but conditions and forecasts are gone. In that display space, the program now displays: 

"Could not retrieve weather data. Is your API key valid? Request failed with status code 401"

I checked all of my API keys, tried a few alternatives, and then plugged in the original API key from tomorrow.io.

In looking at the API calls graph on tomorrow.io, I see that it was consistently pulling 170-ish calls every day, then it dropped to 86 suddenly for one day, then 86 the next day, 4 the next day, and then 0 and that's where it has been for several days.

Any suggestions?

Thanks,
Harry

halifax57, 2022-07-06:
I found this documented problem at tomorrow.io that is parallel in time with my problem, yet I cannot get API key functioning back:

Jul 4, 2022
High Error Rate on Timeline API
Resolved - This incident has been resolved and all API keys are working as expected. If you continue to experience any errors please be in touch with our team at support@tomorrow.io. We will of course continue to monitor all API behavior and we appreciate your patience as we implemented a fix for this.
Jul 4, 17:51 UTC
Update - A fix has been implemented and we are seeing a decrease in errors. We are continuing to monitor the events until we reach an all-clear status.
Jul 4, 16:32 UTC
Identified - The issue has been identified and the team is working on deploying the fix.
Jul 4, 16:17 UTC
Investigating - We are currently investigating an issue impacting our timeline API. 
API calls are currently returning a high volume of 401 and 301 errors. Our team is working diligently to investigate this and we expect a fix to be implemented shortly. We will continue to update accordingly.
Jul 4, 16:03 UTC

[image]

k5map, 2022-07-06:
halifax57,

Am I correct in your page shows the weather data in the right pane or do you have a blank page with the error about retrieving the map?

halifax57, 2022-07-06:
Thanks for the reply, I very much appreciate it. On my display, I have a blank page with the error about retrieving weather data on the right side. See the attached image. I do have a support ticket with tomorrow.io ...

[image]

k5map, 2022-07-07:
Well fudge... we're both having different problems.  If you have time, could you please provide the version of Node & NPM you have installed?  When I check the Mapbox site, it shows that my RPi is not making any API calls so I'm trying to see if the issue with related to a specific version of Node/NPM.
[image]

halifax57, 2022-07-07:
OK. Maybe you have two independent issues?
As requested:
NPM version 6.14.17
Node version v14.19.3
Thank you!

fcrwx, 2022-07-07:
I am seeing the same issue and got it working with the following steps:

1. created a free API key at https://ipstack.com/product
2. modified `server/geolocationCtrl.js`, changing the GET call from `"https://freegeoip.app/json/"` to `"http://api.ipstack.com/{my_ip_address}?access_key={my_api_key}"` 
3. restarted the server (`npm start`)

Hope that helps

BTW, I am on node v12.22.12 and npm 6.14.16

k5map, 2022-07-07:
fcrwx,

Since you've dug into the code, maybe you can answer a question of mine...  we obtain keys from Mapbox and Tomorrow.io but on the webpage in the bottom left corner, it shows Leaflet & RainViewer.  Any idea what relationship those sites have to this app?

halifax57, 2022-07-08:
No joy with my weather-side issue. I'm glad that k5map got his problem fixed, though! Thanks for looking into this.

tomorrow.io support said that they tested my API key and it worked for them. So I am at a complete loss as to why I cannot get this working again.

mlcampbe, 2022-07-08:
halifax57, can you confirm that your pi can pull the data manually using curl? For example, try the following and see if you get data or an error.

curl -v "https://api.tomorrow.io/v4/timelines?location=34.57,86.84&fields=temperature,precipitationProbability,precipitationIntensity,windSpeed&timesteps=1h&apikey=<your-key>&endTime=2022-07-09T12:14:32.193Z"

be sure to replace <your-key> with your actual weather-api key and you will likely need to adjust the endTime value too. This should be the same command that the app is trying to fetch. In my case I get 24 1-hour intervals of data all wrapped into a json string.

This will at least prove if the problem is with the app itself or with your api key (or perhaps a networking error).

halifax57, 2022-07-08:
Hi Mike,
Thanks for the help. Here is the output from entering your curl command, using my API key where XXXXXXXXXXXX is below. I tried it with an endTime of 18:32Z too and got the same result. What does this tell you?
Thanks again!

$ curl -v "https://api.tomorrow.io/v4/timelines?location=34.57,86.84&fields=temperature,precipitationProbability,precipitationIntensity,windSpeed&timesteps=1h&apikey=<XXXXXXXXXXXXX>&endTime=2022-07-09T12:14:32.193Z"

*   Trying 104.18.29.42:443...
* Connected to api.tomorrow.io (104.18.29.42) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=Cloudflare, Inc.; CN=sni.cloudflaressl.com
*  start date: May 27 00:00:00 2022 GMT
*  expire date: May 27 23:59:59 2023 GMT
*  subjectAltName: host "api.tomorrow.io" matched cert's "api.tomorrow.io"
*  issuer: C=US; O=Cloudflare, Inc.; CN=Cloudflare Inc ECC CA-3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x10b3d50)
> GET /v4/timelines?location=34.57,86.84&fields=temperature,precipitationProbability,precipitationIntensity,windSpeed&timesteps=1h&apikey=<nZIX2cNocdmqtXYEpbTW1C4GLt1NmXrM>&endTime=2022-07-09T12:14:32.193Z HTTP/2
> Host: api.tomorrow.io
> user-agent: curl/7.74.0
> accept: */*
>
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* Connection state changed (MAX_CONCURRENT_STREAMS == 256)!
< HTTP/2 401
< date: Fri, 08 Jul 2022 13:55:51 GMT
< content-type: application/json; charset=utf-8
< content-length: 124
< vary: Origin
< access-control-allow-origin: *
< x-moesif-transaction-id: 1fe2f9a6-d38f-4656-84c1-33f08b35e87d
< x-kong-response-latency: 3
< via: 1.1 google
< cf-cache-status: DYNAMIC
< expect-ct: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
< server: cloudflare
< cf-ray: 727951e81f6c59f2-IAD
<
* Connection #0 to host api.tomorrow.io left intact
{"code":401001,"type":"Invalid Key","message":"The method requires authentication, but it was not presented or is invalid."}

mlcampbe, 2022-07-08:
The last line "Invalid Key" seems to be the problem to me. The app is sending off this same command to retrieve the weather data and I suspect it fails the same way too. Therefore I would ignore the app right now and focus on getting the curl command to work. 

The network connection looks ok as it made the connection to api.tomorrow.io successfully. The problem is that the tomorrow.io is rejecting your api-key for some reason. Are there any special characters in it such as * or & or ( that might be getting interpreted?

You might try logging into https://app.tomorrow.io/development/keys and verifying the key you are using is what is shown there. Maybe try to create a new api key?

halifax57, 2022-07-08:
Mike,
When I enter the curl command from tomorrow.api on my key page directly into the console, it works:

curl --request GET --url \
'https://api.tomorrow.io/v4/timelines?location=-73.98529171943665,40.75872069597532&fields=temperature&timesteps=1h&units=metric&apikey=xxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
{"data":{"timelines":[{"timestep":"1h","endTime":"2022-07-13T02:00:00Z","startTime":"2022-07-08T14:00:00Z","intervals":[{"startTime":"2022-07-08T14:00:00Z","values":{"temperature":-41.16}},{"startTime":"2022-07-08T15:00:00Z","values":{"temperature":-41.55}},{"startTime":"2022-07-08T16:00:00Z","values":{"temperature":-42.13}},{"startTime":"2022-07-08T17:00:00Z","values":{"temperature":-43.01}},{"startTime":"2022-07-08T18:00:00Z","values":{"temperature":-43.49}},{"startTime":"2022-07-08T19:00:00Z","values":{"temperature":-43.58}},{"startTime":"2022-07-08T20:00:00Z","values":{"temperature":-43.51}},{"startTime":"2022-07-08T21:00:00Z","values":{"temperature":-43.72}},{"startTime":"2022-07-08T22:00:00Z","values":{"temperature":-43.83}},{"startTime":"2022-07-08T23:00:00Z","values":{"temperature":-44}},{"startTime":"2022-07-09T00:00:00Z","values":{"temperature":-44.15}},{"startTime":"2022-07-09T01:00:00Z","values":{"temperature":-44.95}},{"startTime":"2022-07-09T02:00:00Z","values":{"temperature":-45.8}},{"startTime":"2022-07-09T03:00:00Z","values":{"temperature":-46.07}},{"startTime":"2022-07-09T04:00:00Z","values":{"temperature":-46.06}},{"startTime":"2022-07-09T05:00:00Z","values":{"temperature":-46.51}},{"startTime":"2022-07-09T06:00:00Z","values":{"temperature":-46.37}},{"startTime":"2022-07-09T07:00:00Z","values":{"temperature":-46.35}},{"startTime":"2022-07-09T08:00:00Z","values":{"temperature":-46.66}},{"startTime":"2022-07-09T09:00:00Z","values":{"temperature":-47.17}},{"startTime":"2022-07-09T10:00:00Z","values":{"temperature":-47.77}},{"startTime":"2022-07-09T11:00:00Z","values":{"temperature":-47.92}},{"startTime":"2022-07-09T12:00:00Z","values":{"temperature":-48.09}},{"startTime":"2022-07-09T13:00:00Z","values":{"temperature":-48.16}},{"startTime":"2022-07-09T14:00:00Z","values":{"temperature":-47.41}},{"startTime":"2022-07-09T15:00:00Z","values":{"temperature":-47.27}},{"startTime":"2022-07-09T16:00:00Z","values":{"temperature":-48.21}},{"startTime":"2022-07-09T17:00:00Z","values":{"temperature":-48.19}},{"startTime":"2022-07-09T18:00:00Z","values":{"temperature":-49.32}},{"startTime":"2022-07-09T19:00:00Z","values":{"temperature":-49.98}},{"startTime":"2022-07-09T20:00:00Z","values":{"temperature":-50.31}},{"startTime":"2022-07-09T21:00:00Z","values":{"temperature":-50.93}},{"startTime":"2022-07-09T22:00:00Z","values":{"temperature":-51.37}},{"startTime":"2022-07-09T23:00:00Z","values":{"temperature":-51.32}},{"startTime":"2022-07-10T00:00:00Z","values":{"temperature":-51.18}},{"startTime":"2022-07-10T01:00:00Z","values":{"temperature":-51.02}},{"startTime":"2022-07-10T02:00:00Z","values":{"temperature":-50.67}},{"startTime":"2022-07-10T03:00:00Z","values":{"temperature":-50.44}},{"startTime":"2022-07-10T04:00:00Z","values":{"temperature":-50.2}},{"startTime":"2022-07-10T05:00:00Z","values":{"temperature":-50.03}},{"startTime":"2022-07-10T06:00:00Z","values":{"temperature":-49.88}},{"startTime":"2022-07-10T07:00:00Z","values":{"temperature":-49.71}},{"startTime":"2022-07-10T08:00:00Z","values":{"temperature":-49.64}},{"startTime":"2022-07-10T09:00:00Z","values":{"temperature":-49.94}},{"startTime":"2022-07-10T10:00:00Z","values":{"temperature":-50.4}},{"startTime":"2022-07-10T11:00:00Z","values":{"temperature":-50.93}},{"startTime":"2022-07-10T12:00:00Z","values":{"temperature":-51.36}},{"startTime":"2022-07-10T13:00:00Z","values":{"temperature":-51.77}},{"startTime":"2022-07-10T14:00:00Z","values":{"temperature":-52.18}},{"startTime":"2022-07-10T15:00:00Z","values":{"temperature":-52.63}},{"startTime":"2022-07-10T16:00:00Z","values":{"temperature":-53.09}},{"startTime":"2022-07-10T17:00:00Z","values":{"temperature":-53.52}},{"startTime":"2022-07-10T18:00:00Z","values":{"temperature":-53.92}},{"startTime":"2022-07-10T19:00:00Z","values":{"temperature":-54.25}},{"startTime":"2022-07-10T20:00:00Z","values":{"temperature":-54.53}},{"startTime":"2022-07-10T21:00:00Z","values":{"temperature":-54.68}},{"startTime":"2022-07-10T22:00:00Z","values":{"temperature":-54.86}},{"startTime":"2022-07-10T23:00:00Z","values":{"temperature":-54.8}},{"startTime":"2022-07-11T00:00:00Z","values":{"temperature":-54.79}},{"startTime":"2022-07-11T01:00:00Z","values":{"temperature":-54.83}},{"startTime":"2022-07-11T02:00:00Z","values":{"temperature":-54.69}},{"startTime":"2022-07-11T03:00:00Z","values":{"temperature":-54.44}},{"startTime":"2022-07-11T04:00:00Z","values":{"temperature":-54.32}},{"startTime":"2022-07-11T05:00:00Z","values":{"temperature":-54.47}},{"startTime":"2022-07-11T06:00:00Z","values":{"temperature":-54.42}},{"startTime":"2022-07-11T07:00:00Z","values":{"temperature":-54.55}},{"startTime":"2022-07-11T08:00:00Z","values":{"temperature":-54.48}},{"startTime":"2022-07-11T09:00:00Z","values":{"temperature":-54.61}},{"startTime":"2022-07-11T10:00:00Z","values":{"temperature":-54.6}},{"startTime":"2022-07-11T11:00:00Z","values":{"temperature":-54.8}},{"startTime":"2022-07-11T12:00:00Z","values":{"temperature":-54.79}},{"startTime":"2022-07-11T13:00:00Z","values":{"temperature":-54.88}},{"startTime":"2022-07-11T14:00:00Z","values":{"temperature":-55.46}},{"startTime":"2022-07-11T15:00:00Z","values":{"temp [... cut ...] tartTime":"2022-07-12T21:00:00Z","values":{"temperature":-48.75}},{"startTime":"2022-07-12T22:00:00Z","values":{"temperature":-48.85}},{"startTime":"2022-07-12T23:00:00Z","values":{"temperature":-48.76}},{"startTime":"2022-07-13T00:00:00Z","values":{"temperature":-48.95}},{"startTime":"2022-07-13T01:00:00Z","values":{"temperature":-49.32}},{"startTime":"2022-07-13T02:00:00Z","values":{"temperature":-49.15}}]}]}}

...but not in the app. Should I do a complete refresh of the program?
Kind regards!

mlcampbe, 2022-07-08:
That is interesting. Are you running my curl command the the one from the webgate both from the same place (the pi)?

halifax57, 2022-07-08:
Yes, from the same command line on the same pi.
Thank you again.

halifax57, 2022-07-08:
Thanks, Mike. I appreciate your help which demonstrated what was working. Given my experience level debugging Linux, I will likely go back and start with a fresh OS and rebuild from scratch and see where that takes me.
Kind regards!

lerilar, 2022-07-10:
Hope you'll find an answer @halifax57 - the same 401 issue surfaced here as well :( API key curl from the tomorrow.io site works fine here too.

mlcampbe, 2022-07-10:
Have you edited the project to change the references of climacell to tomorrow.io? The files in the project still use climacell which may redirect to tomorrow.io but perhaps is not fully equivalent?

lerilar, 2022-07-11:
I have not edited anything - and that part worked fine until yesterday actually (when I updated the pi and rebooted). I see outgoing calls to data.climacell.co, but when I check in the browser they are redirected to tomorrow.io and tomorrow.io shows activity on that token at least. I have no idea which call is causing the 401 message to appear however. This is what it looks like:

[image]
.

RNET1, 2022-07-12:
Have you tried entering the LAT LON values? I've had the same issues, but populating those fields and all is well again.

halifax57, 2022-07-12:
> Have you edited the project to change the references of climacell to tomorrow.io? The files in the project still use climacell which may redirect to tomorrow.io but perhaps is not fully equivalent?

I have always used the tomorrow.io API key. It worked fine for months until tomorrow.io's July outage. Nothing since. Thanks.

halifax57, 2022-07-12:
> I have not edited anything - and that part worked fine until yesterday actually (when I updated the pi and rebooted). I see outgoing calls to data.climacell.co, but when I check in the browser they [...]

That is exactly what my system has looked like since July 4th tomorrow.io outage. Radar and mapping fine, no wx data and it reports the error w/comment just like in your photo. Can't find my way back! I will be doing a build from scratch shortly as I do not have the depth of talent in this realm to debug competently. Of course, I am hoping somebody finds a fix in the meantime.

RNET1, 2022-07-13:
I recommend the easy test, unless you have already tried this... populate the LAT and LON in the settings * just to see if that makes a difference. I had odd behavior too -  tried all sorts of tests, including new builds and even going back to V1 of the app. In the end, populating those LAT/LON values resulted in normal data display.

halifax57, 2022-07-13:
> I recommend the easy test, unless you have already tried this... populate the LAT and LON in the settings * just to see if that makes a difference. I had odd behavior too - tried all sorts of tests, [...]

RNET1, I already had LAT/LONG data in my settings.json file, but I went in and looked again, then looked at the settings.example.json and saw that he had 4 decimal places for his LAT/LONG x.xxxx. I trimmed my 6-decimal place LAT/LONG to 4 and did a reboot and I am up and running. I rebooted a few times and all still good. 

Thank you, kind friend, for the help! Yay!

lerilar, 2022-07-13:
I did the exact same thing - trimmed to 4 decimals and boom it started working again. I still don't have the map showing anything but grey, but as long as I see the rainviewer I am happy. Thanks for all the help.

chrism583, 2022-07-14:
Editing my Lat/Lon in settings.json also resolved my issue of not seeing weather information.  Thanks!!!

MrGingerT, 2022-07-14:
Just to confirm entering the lant and long does fix the blank screen issue fully. No idea why though. Many Thanks to those who found this out.

k5map, 2022-07-15:
@fcrwx 

When you modified the JS files in the SRC tree, did those changes take effect immediately or did you have to enter npm install or some other command before the changes would take effect?

RNET1, 2022-07-15:
If you modify the source files, then a rebuild is required:
Replace data.climacell.co with api.tomorrow.io -- Change in the client/src/AppContext.js file
(Note - there are a few references to change here, search for climacell to find them)

Then rebuild the dist/bundle.min.js ... run the following
cd client 
npm run-script prod
(The following is ONLY If you are using nodejs v17 (it will fail with a "hash" error). 
However, it must be run prior to the npm command  
export NODE_OPTIONS=--openssl-legacy-provider

If you get any errors, read the output - it has directions to solve. Otherwise, post here for help.

bnelson333, 2023-06-17:
I was also getting an intermittent error on not being able to retrieve weather data but my error was error 429 not 401. I trimmed the lat/lon in settings.json per this thread from 6 decimal places to 4 and it also fixed my issue. Would the developer be able to implement a quick round function on that so others don't run into the same issue? (sorry I have no idea how to contribute to a project like this)

pbhound, 2026-08-25:
so i got this working but would like it to auto start when powered on how do i accomplish this?