elewin/pi-weather-station #38: Error 403 for 7-day weather forecast

mf1x, 2023-05-18:
Hi,

For a couple of weeks now I have been getting the following error message where the 7-day forecast should be:

"Cannot get 7 day weather forecast
Request failed with status code 403"

Has the weather provider changed something?

Regards
Max

k5map, 2023-05-18:
I'm experiencing the same problem but don't know when it started.

ZeroCool-Dade, 2023-05-20:
I have the same issue, wondering if its related to the tomorrow.io free account only gets a 5 day forecast?

mf1x, 2023-05-20:
Could be. Have they changed that recently?

k5map, 2023-05-21:
I just logged into my Tomorrow.io account and it shows no calls for my account.  Anyone know if the current weather is coming from Tomorrow or Mapbox?

klamer, 2023-05-23:
I'm so glad I'm not the only one.  Been looking for this issue for at least a week or two.  I'm making the API call into tomorrow.io but just getting the 24 hour.
[image]

mf1x, 2023-05-24:
Let's hope someone can help :D

klamer, 2023-05-27:
I messaged him on reddit, I'm hoping it's an easy fix...<shrug> I'm no coder.

k5map, 2023-06-04:
@jbeckley11 ... can't thank you enough for figuring this out issue and coming up with the fix.

jbeckley11, 2023-06-04:
@k5map To rebuild the project, navigate into the client folder and do `npm run prod`
Then go back to the app directory and do `npm start`

k5map, 2023-06-04:
@jbeckley11 ... thanks for the commands.  When I run "npm run prod", I get this error:

pi@pi-Weather:~/pi-weather-station/client $ npm run prod

> pi-weather-station-client@1.0.0 prod
> webpack -p --env.BUILD_PRODUCTION=1

sh: 1: webpack: not found
npm ERR! code 127
npm ERR! path /home/pi/pi-weather-station/client
npm ERR! command failed
npm ERR! command sh -c webpack -p --env.BUILD_PRODUCTION=1

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2023-06-04T11_48_42_756Z-debug.log

Any ideas what could be wrong?

ZeroCool-Dade, 2023-06-04:
@k5map   not sure if its the correct way to do it but this is how I got around the webpack error.. :
from the client folder run 
sudo npm install
sudo npm run prod

k5map, 2023-06-04:
@ZeroCool-Dade ... that did it, Much Thanks!

elewin, 2023-06-05:
#39 
7 day forecast changed to 5 day forecast to meet new API requirements

closed 2023-06-05 by elewin

elewin, 2023-06-05:
> @elewin... this is not a big deal but did you also update file client/src/components/weatherCharts/DailyCharts/index.js line 30 from 7 Day to 5 Day?

Yep!

https://github.com/elewin/pi-weather-station/pull/39/commits/da05477d54831834e1fc600f6ed6af68c0c4038a

mf1x, 2023-06-06:
@elewin Are you able to merge the fix with Sean Riggs's Docker fork?

klamer, 2023-06-09:
Thanks all for their help in resolving this and @elewin for updating the code.

mf1x, 2023-06-09:
Yep. Thanks a lot guys :)

duarch, 2023-06-20:
What about docker image?

mf1x, 2023-06-20:
Not sure. I haven't gotten any response regarding this topic. I guess you need to do change it yourself for now.

Install nano inside the container and change it (it's Alpine Linux btw).

IamPatticus, 2025-04-03:
I can't seem to get the docker image to reflect the changes to the AppContext.js and Index.js ....not a big deal but would like to see the error go away :)