elewin/pi-weather-station #42: Cannot see map

Forensic-GUY, 2023-07-01:
I can see all but the map, I verified my API's are correct mp MapBox, tomorrow.ai and LocationIQ.

Photo: 

[image]

crwdplezer, 2023-08-16:
> 

I had to change the lat and long settings to xx.xxxx and it fixed it. If it has more than 4 #'s after the decimal then it was causing issues..same for the weather data

Forensic-GUY, 2023-08-16:
> > 
> 
> I had to change the lat and long settings to xx.xxxx and it fixed it. If it has more than 4 #'s after the decimal then it was causing issues..same for the weather data

I had it set to 4 decimal points already but still wont load

chwiejak, 2023-10-16:
I also cannot see the map, please let me know how to change lat and long to xx.xxxx, my is xx.xxxxxxxxxxxxxxxxxxxxxxxxxx