elewin/pi-weather-station #44: 5 Day Forecast on docker

Skunk-Cabbage, 2023-08-26:
Has anyone figured out how to get the 5 day forecast to work again on the docker container?  I've tried editing the container and rebuilding it with new config, but no luck so far.

mf1x, 2023-09-07:
I just edited the file inside the container and restarted the container. I didn't rebuild it.

grenius, 2023-10-15:
@mf1x where/what did you have to edit? I have map and weather but no forecast or telemetry.

mf1x, 2023-10-15:
@grenius
In https://github.com/elewin/pi-weather-station/issues/38 jbeckley11 provided a solution: https://github.com/elewin/pi-weather-station/issues/38#issuecomment-1575432909

grenius, 2023-10-15:
@mf1x thanks dude, I changed that 6 to a 5 and rebuilt the container but I'm still not seeing it. Hmm.

mf1x, 2023-10-15:
Try stopping the container, change the file and start it again. Don't rebuild it.

grenius, 2023-10-16:
@mf1x the issue is I didn't have access to an actual PC and mobile browser wasn't rendering it. Thanks again.

mf1x, 2023-10-16:
Np.