elewin/pi-weather-station #54: Action Required: Migrate Your Tomorrow.io API Domain

thicla01, 2024-04-28:
Hello,

I've received this communication from tomorrow.io team. I supposed some modifications will be needed. Will you update your project to adapt?

Regards,

     

> _Hi there,_
> 
> _We're reaching out with an important update regarding your API services with Tomorrow.io._
> 
> _As part of our ongoing efforts to improve service quality, integrity, and security, we are migrating our API domain._ 
> 
> _Upon reviewing our records, we noticed that your account is currently utilizing the old API domain._
>  
> _To avoid any interruption in service, we kindly ask you to update your API endpoint from data.climacell.co to  api.tomorrow.io  by May 11._
> 
> _If you have any questions or need assistance with the migration process, please feel free to reach out to our customer success team at [support@tomorrow.io](mailto:support@tomorrow.io)._
>  
> _We are here to support you every step of the way and ensure a seamless transition to the new API domain._
>  
> _Thanks,_ 
> _The Tomorrow.io Team_

chrism583, 2024-05-06:
Any progress on this?  Are we able to just locate uses of data.climacell.co and change them to api.tomorrow.io?

gdaysoul, 2024-05-09:
No news? I tried finding the locations where the changes are needed but I haven't found the files.  

I did find in  Opeweather option? #23  some way of changing things but as a "rookie" some of the details were needing more steps I think.

aevans1987, 2024-06-12:
Hey All, I forked this project and make the appropriate changes:

https://github.com/aevans1987/pi-weather-station

Gonna open a pull request to get it updated here, but no idea if it will be accepted.