elewin/pi-weather-station #63: Initial Zoom Functionality

eights-dev, 2025-01-10:
Wasn't sure where to ask this so feel free to remove if not appropriate, but is there a way to set the initial zoom level? I'm running this on a pi remotely via a browser and when it loads its zoomed in on my location but I was looking for a way to have the initial view zoomed out two or three times. I've looked through the config and settings files but didn't seen anything. Any help or advise is greatly appreciated.

mlcampbe, 2025-01-10:
Take a look at https://github.com/elewin/pi-weather-station/issues/28 and adjust the zoom level and rebuild the scripts. I think if you want to zoom out you would use a number smaller than the default value of 9.

eights-dev, 2025-01-10:
> Take a look at #28 and adjust the zoom level and rebuild the scripts. I think if you want to zoom out you would use a number smaller than the default value of 9.

Ahh ok, perfect. I'll dig through it and change the default value of 9 to say 5 and see what happens. Thank you!!!!!!

eights-dev, 2025-01-10:
Closing the "issue", the change you suggested worked like a charm. Thanks again!

closed 2025-01-10 by eights-dev