elewin/pi-weather-station #66: Could not retrieve weather data

mf1x, 2025-04-26:
Hi,

Is anyone else experiencing the following?

"Could not retrieve weather data.
Is your weather API key valid?

Network Error"

[image]

Started two days ago. Re-checked the tomorrow.io key and also created a new one.

Have they changed something?

Regards
Max

k5map, 2025-04-26:
Ditto for me and another friend who has this app running in a RPi4.

mlcampbe, 2025-04-26:
Yes I had the same problem. You need to edit the client/src/AppContext.js and change the host name data.climacell.co to api.tomortow.io and then rebuild the js files using this:

$ cd client
$ export NODE_OPTIONS=--openssl-legacy-provider
$ npm run prod

It should complete with some warnings which is ok. Then restart the app and it should work.

k5map, 2025-04-26:
@mlcampbe ... I followed the steps in your post but still do not get the data in the right pane.  Is there more than one reference to climacell in the js file?

k5map, 2025-04-26:
@mlcampbe ... ok, my bad... I found multiple entries.  Even after updating all of the entries, I still do not get the data displayed.  Also, when I ping the name you posted (api.tomortow.io), it fails to find the host.  Is that the correct host name?

mf1x, 2025-04-26:
@mlcampbe Thanks a lot :) I'll try it later.

@k5map Probably a typo. Pretty sure they meant "api.tomorrow.io".

k5map, 2025-04-26:
@mf1x @mlcampbe ... I just updated my js file to use api.tomorrow.io and still no data being displayed

mf1x, 2025-04-26:
Currently, I run the Docker container, and "npm run prod" says "--openssl-legacy-provider" is not allowed.

I'll set it up in an LXC later and let you know if I was able to make it work.

chwiejak, 2025-04-26:
> Yes I had the same problem. You need to edit the client/src/AppContext.js and change the host name data.climacell.co to api.tomortow.io and then rebuild the js files using this:
>
> $ cd client $ [...]

Hi, I changed all three "data.climcell.co" to "api.tomorrow.io" (only this, not whole links), made those commends, got: 
sh: 1: webpack: not found

and weather still not working:(

peter279k, 2025-04-27:
I also encounter this issue and my solutions are as follows:

1. Replacing the `data.climcell.co` with the `api.tomorrow.io`. And it will do three times.
2. Running the `cd client/`
3. Running the `npm install --save-dev webpack` command to install the webpack tool.
4. Running `npm run prod` command to build the client for pi-weather-station. And the running command log is as follows:

```sh
$ npm run prod

> pi-weather-station-client@1.0.0 prod /home/lee/pi-weather-station/client
> webpack -p --env.BUILD_PRODUCTION=1

Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
Hash: cac031869a554338b0fa
Version: webpack 4.47.0
Time: 46780ms
Built at: 04/28/2025 12:11:53 AM
          Asset       Size  Chunks                    Chunk Names
   ./index.html  743 bytes          [emitted]
1.bundle.min.js  388 bytes       1  [emitted]
  bundle.min.js    904 KiB       0  [emitted]  [big]  main
Entrypoint main [big] = bundle.min.js
 [38] (webpack)/buildin/module.js 497 bytes {0} [built]
[270] ./node_modules/moment/locale sync ^\.\/.*$ 3.16 KiB {0} [optional] [built]
[284] (webpack)/buildin/global.js 472 bytes {0} [built]
[371] ./src/styles/index.js 53 bytes {0} [built]
[372] ./src/index.js + 76 modules 244 KiB {0} [built] [2 warnings]
      | ./src/index.js 322 bytes [built]
      | ./src/AppContext.js 22.1 KiB [built]
      | ./src/settings.js 319 bytes [built]
      | ./src/services/geolocation.js 1.2 KiB [built]
      | ./src/services/reverseGeocode.js 1.29 KiB [built]
      | ./src/services/conversions.js 2.14 KiB [built]
      |     + 71 hidden modules
[373] ./node_modules/style-loader/dist/cjs.js!./node_modules/css-loader/dist/cjs.js!./src/styles/main.css 524 bytes {1} [built]
[376] ./node_modules/css-loader/dist/cjs.js!./src/styles/main.css 342 bytes {1} [built]
    + 596 hidden modules

WARNING in ./src/components/App/index.js
Module Warning (from ./node_modules/eslint-loader/dist/cjs.js):

/home/lee/pi-weather-station/client/src/components/App/index.js
  29:6  warning  React Hook useEffect has missing dependencies: 'getBrowserGeo', 'getCustomLatLon', and 'loadStoredData'. Either include them or remove the dependency array  react-hooks/exhaustive-deps

✖ 1 problem (0 errors, 1 warning)

 @ ./src/index.js 3:0-35 6:109-112

WARNING in ./src/components/WeatherMap/index.js
Module Warning (from ./node_modules/eslint-loader/dist/cjs.js):

/home/lee/pi-weather-station/client/src/components/WeatherMap/index.js
  38:27  warning  React Hook useCallback received a function whose dependencies are unknown. Pass an inline function instead  react-hooks/exhaustive-deps

✖ 1 problem (0 errors, 1 warning)

 @ ./src/components/App/index.js 4:0-49 35:38-48
 @ ./src/index.js

WARNING in asset size limit: The following asset(s) exceed the recommended size limit (244 KiB).
This can impact web performance.
Assets:
  bundle.min.js (904 KiB)

WARNING in entrypoint size limit: The following entrypoint(s) combined asset size exceeds the recommended limit (244 KiB). This can impact web performance.
Entrypoints:
  main (904 KiB)
      bundle.min.js

Child HtmlWebpackCompiler:
     1 asset
    Entrypoint HtmlWebpackPlugin_0 = __child-HtmlWebpackPlugin_0
    [0] ./node_modules/html-webpack-plugin/lib/loader.js!./src/index.html 774 bytes {0} [built]
```

5. Done. Restart the server and this issue will be resolved.

After completing above steps, I also found that `https://api.tomorrow.io/v4/timelines?location=${latitude}%2C${longitude}&fields=${fields}&timesteps=current&apikey=${weatherApiKey}` has the problem.

It seems that the Tomorrow API with timesteps `current` is not worked because the response data will be empty.

To fix this issue, I replace the `current` with `1h` and it will be fixed temporarily.

fredt152, 2025-04-27:
Was having the same here, now fixed for me:

- replaced the required climacell entries for tomorrow.io in client/src/AppContext.js

- cd back to client dir, export NODE_OPTIONS=--openssl-legacy-provider, npm run prod

when running this ^^^ I was getting an error

> pi-weather-station-client@1.0.0 prod
> webpack -p --env.BUILD_PRODUCTION=1

[webpack-cli] Error: Unknown option '-p'

- switched out the following lines at the top of client/package.json

 "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1",
    "dev": "webpack --progress --watch --mode development",
    "prod": "webpack --progress --watch --mode production"
  },

- npm install

- npm run prod

- cd back to parent dir 

- npm start

hope this helps

wayoutthere007, 2025-04-27:
Still not working for me after following the last two solutions. Could someone replace the necessary updated files and those of us less adept could uninstall/reinstall the repository?

mlcampbe, 2025-04-28:
For those that still cannot get the solution working I am going to guess that the javascript bundles may not be getting rebuilt. After editing the AppContext.js file and running "npm run prod" go to the pi-weather-station/client/dist directory and check the date/time of the 3 .js files there. They should all be updated to the current date and time if the rebuild commands worked properly.

k5map, 2025-04-28:
@mlcampbe ... that's exactly my situation... all 3 of my files still have a Dec 2023 date.

wayoutthere007, 2025-04-28:
My files updated according to the time stamps. Still broken.

wayoutthere007, 2025-04-28:
Is there a working alternative to this effort? I really enjoyed my pi weather station running. 
An uninstall, reinstall, fixes applied again, still failed for me so I'm giving up on this distro unfortunately.

wayoutthere007, 2025-04-29:
This is the error I get when running "npm run prod". It breaks the html portion and then doesn't complete. I have to CTRL+C out.

[image]

peter279k, 2025-04-29:
> This is the error I get when running "npm run prod". It breaks the html portion and then doesn't complete. I have to CTRL+C out.
> 
> [image]

Which Node.js version do you use?

k5map, 2025-04-29:
When I run the commands outlined by @mlcampbe , I get this response:

pi@pi-Weather:~/pi-weather-station/client$ export NODE_OPTIONS=--openssl-legacy-provider
pi@pi-Weather:~/pi-weather-station/client$ npm run prod

> pi-weather-station-client@1.0.0 prod
> webpack -p --env.BUILD_PRODUCTION=1

[webpack-cli] Error: Unknown option '-p'
pi@pi-Weather:~/pi-weather-station/client$

rizzl3r, 2025-04-29:
> Was having the same here, now fixed for me:
>
> * replaced the required climacell entries for tomorrow.io in client/src/AppContext.js
> * cd back to client dir, export [...]

This fixed it for me, thanks @fredt152  - had to undo some of my other attempts via `git checkout .` from the parent directory. After that, followed these, let it do it's magic and it's back to the races.

k5map, 2025-04-29:
@rizzl3r ... much thanks for the instructions.  After following the steps in your post, I am now able to view the data on the right side of the screen.

wayoutthere007, 2025-04-29:
LOL - so k4map thanks @rizzl3r  who thanks @fredt152 for thier steps. the stupidity of it all. it'd have been preferred if you would have documented your exact steps to your resolution - or say "mines working thanks". LULZ

So, in my case, I'm unable to run npm at all after doing the "export NODE_OPTIONS..." line

[image]

peter279k, 2025-04-29:
@wayoutthere007, could you unset the `NODE_OPTIONS` then run the `npm run prod` command directly?

Brooks09211998, 2025-04-29:
I followed the steps that @fredt152 outlined, im still not seeing anything pop up on the side pane.

wayoutthere007, 2025-04-29:
unsetting NODE_OPTIONS allowed npm to run, however it stuck and never completed. I'm rebuilding from scratch. Pi OS is installed, getting vnc running, then we'll try again from scratch. 

Thanks for the help guys, I appreciate it.

wayoutthere007, 2025-04-30:
UPDATE - here's how I was able to get mine working.
- rebuilt my pi with Debian Bullseye (for builtin REALVNC x11 access) using Pi's imaging tool offered on their site
- sudo apt update   and then  sudo apt upgrade
- sudo apt install firefox-esr
- git clone https://github.com/elewin/pi-weather-station
- cd pi-weather-station
- curl -sL [https://deb.nodesource.com/setup_14.x | sudo bash -
- sudo app install nodejs
NOW FIX 2 FILES
- replaced the required climacell entries for api.tomorrow.io in client/src/AppContext.js and save
- cd back to client
- update package.json prod line with: "prod": "webpack --progress --watch --mode production" and save
- run npm install
- run npm run prod (this will hang at the end - it's ok to ctrl+c out of it)
- reboot
- open a terminal and go back to pi-weather-station
- run npm install
- run npm start

This is how my right side started working again after it launched in firefox and I set my keys!! 

So thankful for the insight given by k5map (package.json update) and fredt152 (appcontext.js update).

I did not have to do the NODE_OPTIONS for mine to work.

Thanks again guys! I've made a disk image of my current setup in case my darn microSD goes tits up. 

ps. I followed this build which is why i used nodejs v14.x.  https://www.youtube.com/watch?v=cTGTB1sD3MQ

[image]

mf1x, 2025-05-01:
Great, works for me now. Thanks everyone. I'll close this.

I hope this will make it into the code at some point :D

closed 2025-05-01 by mf1x

new2brew, 2025-05-04:
Thanks to all for the input, but the problem for me is (and I admit I am not an expert on any of this) there are so many attempts and fixes that I am confused. I tried to follow wayouthere007's steps, but I don't feel that they are complete (could be wrong).. I found I had to do: the npm install --save -dev webpack before npm run prod would work. Also ran into a memory issue as I was trying to fix on a Zero 2 W which just further complicated everything. Retrying on a Raspberry 4 did not seem to work, so I am back at square one. I guess my request is this -- Is there a proven series of steps to follow? Not a complaint... I appreciate the help.... just confused. Thanks

wayoutthere007, 2025-05-04:
I know my set of steps worked for me on my Pi 3 B. I did update my "npm install" to be "npm install --save --dev webpack" in case other installs needed it. I'm not sure about the memory error as I have the Pi 3 B. If you have the option on Pi 4, try doing a Bullseye repo and not the newer Wayland. I used Bullseye so I know it works. Good luck! Let us know how it goes.

new2brew, 2025-05-05:
Thanks wayoutthere007... I tried the additions to your steps. I am using bookworm?.. I believe Linux 6.12.25+rpt-rpi-v8... could that be an issue?  Bullseye vs Bookworm?  I am not even sure what Wayland is. I have restarted at least 3 times from scratch flashing the SDcard and gone through all the steps, but nothing seems to work... that is, the right hand panel is not working. I feel I am so close. I am using a PI4B. I have not done curl -sL https://deb.nodesource.com/setup_14.x | sudo bash - I get a script depreciation warning so Instead I have sudo apt install npm and nodejs. node -v is v18.19.0
Anyway again I appreciate the input. I will start again tomorrow

wayoutthere007, 2025-05-05:
Even though you get a deprecation warning with nodejs v 14x, it'll still install. It just does a timeout so you can read the warning but will install. Mine did. I went with 14x because I know it works with the code.

new2brew, 2025-05-05:
Thanks to all and especially wayoutthere007 I have finally succeeded. I took it very slow (following wayoutthere's steps) as I tend to get ahead of myself. FYI - running on a Raspberry 4 B but plan to swap the SDcard in to a Zero 2 W and hope it works on that too... it should. It just can't be built on a Zero 2 W (with the revisions) due to memory limits - at least that was my experience. I am using Bullseye Linux 6.1.21-v8+.... I am using 32 bit version as I believed the latest bookworm version was possibly a problem..... it might not be. -- I am not going to research it further LOL In any case thanks again and I will post if I have a problem switching to the Zero 2 W otherwise assume it worked. Wow that was more work than I intended but it's done...

new2brew, 2025-05-05:
Update of last comment from me about running pi-weather-station on Pi Zero. 2 W. Yes it is possible, but extremely slow to load and I'm not sure how well it updates. Running in Chromium. If it's the only thing you want to display or do on a Zero 2 W it might be ok..... time will tell

morphias2004, 2025-05-08:
@wayoutthere007 I think this line needs to be corrected to `sudo apt-get install -y nodejs`

[image]

wayoutthere007, 2025-05-08:
Good eye! Edited.

tehfletch, 2025-05-11:
Can someone clarify the instruction for "replaced the required (3) climacell entries for api.tomorrow.io in client/src/AppContext.js".  I don’t understand what I am supposed to do. I can find the three climacell entries, but I don’t have a clue of what I am supposed to fix.
Thanks for any advice.

new2brew, 2025-05-11:
tehfletch --  Instead of /…climacell…./ (I put the … because I didn’t remember the exact spelling) type in  api.tomorrow.io  instead --- everything between the two //. There are three http entries in the file that need to be done.  Also remember to substitute 1h for current (only) in that file. It’s one of the steps. Hope that clarifies it

tehfletch, 2025-05-12:
Thanks for the reply, I'll try it again.

Sent from my Verizon, Samsung Galaxy smartphone