elewin/pi-weather-station #67: Network Error on Right hand Panel

new2brew, 2025-05-02:
I've recently tried to to install this on two different Raspberry Pi's. One a Raspberry 4 and the other a Zero 2 W. In both installs radar works, etc but I do not get any display in the right panel. and it says it has a Network error. I've double and triple checked the APIs and they all seem to be correct. Although I am not sure whicj API is used for the right panel info... I believe it is the Tomorrow API. I'm wondering if anyone else is having this issue. Is the whole project still working as it should or is it depreciated in some way. I am running PiHole but I disabled that on my network temporarily but that did not seem to make any difference. Any feedback would be appreciated.... I am running the latest and greatest Pi OS 2024-11-19 ---- I believe that is the newest

mschivou5, 2025-05-02:
I had the same temperature for a week, rebooted the pi and now I have this same network issue.

tjdziedzinski, 2025-05-03:
I am also experiencing the same. Double checked my API key as well.

new2brew, 2025-05-03:
Well at least , in my case, I feel a bit better as I now know it’s not user
error. I hope it can be fixed since it is a potentially great little
weather display.  Thanks to all that have responded.

mlcampbe, 2025-05-03:
Take a look at the steps in #66

new2brew, 2025-05-03:
Ok. Thanks I get the gist of it. There are so many posts I’m going to try
and follow wayoutthere007 steps near the end of the thread. Hope it works

morphias2004, 2025-05-08:
Looks like this is the issue. Tomorrow.io won't accept 'current' anymore. Needs to be changed to 1h.

[image]

mlcampbe, 2025-05-08:
What about installing a different browser? I am not sure firefox is lighterweight that chromium. Or, try to get midori installed and test with it.

rexkp, 2025-05-08:
midori works on an original gen Model B, but only just.

new2brew, 2025-05-08:
OMG... thanks to mlcampbe for the suggestion to use Midori... Its worked and it came up on screen fast.... Thanks for the suggestion.... I had to delete chromium so that it wouldn't keep defaulting to using that browser... but once that was done.... bingo!

VelocityAddiction, 2025-05-15:
I've tried everything mentioned above including the changes in #66 but still having an issue with the side bar info not showing.

PaulKTM, 2025-05-16:
Same here, did the API changes to tomorrow last year and all has been working well, this started in the past few days, maybe even under 24 hrs. Only the right side data is affected...

ksr, 2025-06-25:
Hi all, all calls to data.climacell.co need to be changed to api.tomorrow.io.